gkjohnson/urdf-loaders #8: Support more joint types

gkjohnson, 2018-02-03:
http://wiki.ros.org/urdf/XML/joint

`prismatic`, `floating`, `planar`

`<mimic>` tag

EzraBrooks, 2024-02-12:
I'm intending to contribute planar joint support sometime this week, as my project that uses `urdf-loaders` needs to support whole-body motion planning (achieved in the simplest case via an infinitely-variable "planar joint" representing the X,Y location of the robot). Do you have any advice or preferences on how I should implement the API?

gkjohnson, 2024-02-13:
@EzraBrooks Awesome!

I think the `setJointValue` function can just be modified as it is. `URDFJoint.setJointValue` already takes a set of values to accommodate a multi dimensional joint case, so setting the joint would look like so:

```js
robot.setJointValue( 'joint_name', x_value, y_value );

// or

robot.joints.joint_name.setJointValue( x_value, y_value );
```

This, of course, only lets you set both dimensions at a time. Setting one at a time could be done by just reusing the existing set joint value but if you need something more convenient then something like the following interface could work where the first argument is the index into the joint value tuple:

```js
joint.setJointValueComponent( joint_value_index, joint_value );
```

EzraBrooks, 2024-02-15:
thanks. got held up behind another bug so far this week but myself or one of my teammates should be taking a stab at this soon :)

PR #281 (2024-08-28): Add support for floating, planar joint types
Closes #8 

WIP, need to add an example (and debug any issues said example ends up having)
closed 2024-08-28 by gkjohnson