gkjohnson/urdf-loaders #44: Auto centering can cause model to bounce

gkjohnson, 2018-06-25:
Updating the camera focus based on the robot bounding box can cause the model to move up and down too much. See line https://github.com/gkjohnson/urdf-loaders/commit/63f599a78646ee464b2e3103739308459dd3aab6#diff-2a12f3714b8acb49f99007cc2a720c89R238

ipa-jfh, 2018-06-26:
> If disabled in this case, would you just expect it to center about world 0, 0, 0?

I do favour that the viewer initially detects the robot's bounding box and sets centre and focus automatically. 
Maybe you could also provide a function for this, so that I can centre the viewer at certain point of time, when e.g. the first frame of the animation has been loaded

gkjohnson, 2018-09-20:
Fixed in #70

closed 2018-09-20 by gkjohnson