gkjohnson/urdf-loaders #82: Parse the rest of the URDF (Collision, Intertia)

gkjohnson, 2018-10-18:


chaitanya-deep, 2019-02-21:
I wanted to pick this up.
Shall I add `visualEnabled` and `collisionEnabled` in the options similar to RViz?
http://wiki.ros.org/rviz/DisplayTypes/RobotModel#line-9

gkjohnson, 2019-02-22:
Great! What were you imagining the API looking like? I haven't given it a whole lot of thought.

chaitanya-deep, 2019-03-04:
I was thinking of something like this
https://github.com/rapyuta-robotics/urdf-loaders/commit/c3974ab0b043d211f06a857fcc9a1e424f40f321

gkjohnson, 2019-03-04:
I like it being an option to URDFLoader. So you'd load collision geometry like so:
```js
new URDFLoader().load( url, onLoad, null, null, {
    visualEnabled: false,
    collisionEnabled: true,
} );
```

If both `visualEnabled` and `collisionEnabled` are true would it both meshes display on top of each other?

chaitanya-deep, 2019-03-04:
Yes. I initially thought of an option named `geometryType` which could be `visual` or `collision` but followed RViz. I don't have a preference for either. My current usage doesn't require both the meshes.

Yes if they're both true, they should be displayed on top of each other but since collision should ideally encompass the visual geometry, for most use cases user should never see the visual if both are enabled.
In the case of T12, the visual and collision meshes are the same hence this cannot be seen in action.

gkjohnson, 2020-10-10:
Closing in favor of #184, #185

closed 2020-10-10 by gkjohnson