gkjohnson/urdf-loaders #96: Make the root link the same as the robot node

gkjohnson, 2018-12-19:
Right now we're wrapping the root link node in a container object when we could just be making the robot node the same as the root link node.

chaitanya-deep, 2019-01-27:
Is this really necessary? The robot and the root link are two distinct tags in the urdf and the loader makes the same distinct objects of different types `URDFRobot` and `URDFLink` as would be intuitive. Removing `URDFRobot` would be a loss of information.

I'm actually in the process of editing the urdf-loaders urdf-exporter-js to get back the same urdf xml that was loaded by the loader with the api taking no other options from the user. I do make a few assumptions for this - unique names, storing geometry mesh filenames in userData while assuming the user didn't edit the geometry. More importantly I'm only traversing for objects of type `URDFLink`, `URDFJoint` and `URDFRobot`.
If the robot node is removed, the exporter has no way of knowing the name for robot tag in the original urdf

gkjohnson, 2019-02-12:
Hey @chaitanya-deep any further thoughts on this?

I have a PR open at #114 showing what it would look like. The robot name is preserved on `robot.robotName` and the robot xml node is available on `robot.urdfRobotNode`.

Thanks!

chaitanya-deep, 2019-02-20:
Hi,
Sorry I missed the notifications. This will work.
Although it's not backward compatible. I'll update my application to consider this change.
It makes sense that the robot node has nothing apart from name and the name could just be a property of the root link.

gkjohnson, 2019-02-20:
> Although it's not backward compatible.

Yeah unfortunately there are some breaking changes in the upcoming `0.6.0` release but I'll try to have them all documented.

PR #114 (2019-02-22): Make root link into Robot node
Fix #96
closed 2019-02-22 by gkjohnson