gkjohnson/urdf-loaders #119: Wrong pose for a link

ipa-jfh, 2019-02-14:
The drag and drop feature is really great!

I tried it once with this [urdf](http://wiki.ros.org/urdf/Tutorials/Building%20a%20Visual%20Robot%20Model%20with%20URDF%20from%20Scratch) (I removed all `geometry` tags that use meshes). For some reason it shows one of the wheels with a wrong rotation.

### UrdfViewer:
[image]

### Rviz:
[image]

gkjohnson, 2019-02-14:
Interesting! I thought I'd tracked down all the rotation issues. Cylinders are an odd case because two rotations are applied (one to orient the cylinder the way that the URDF spec expects, and one to apply the mesh rotation). I'll take a look in a bit to see what the issue is.

> The drag and drop feature is really great!

> (I removed all `geometry` tags that use meshes)

Thanks! In chrome you can drag multiple files, including the geometry files, and it should load all of them properly. I'm not sure if there's a way to do this in Firefox, at the moment.

ipa-jfh, 2019-02-14:
> In chrome you can drag multiple files, including the geometry files, and it should load all of them properly.

This sound interesting. I think it would be great if users can just go to e.g. the gh-pages instance, drag and drop their urdf, somehow assign the pkg-dependencies and finally see their robot visualized.

Actually, I tried today to add a pop-up that would ask for all missing package-references.

What do you mean by multiple files? One after the other, or a folder? 

I guess by geometry you mean the meshes? How is it possible to reference them correctly with the package-name<>file-path.

Thanks for your quick reply!

gkjohnson, 2019-02-14:
> I think it would be great if users can just go to e.g. the gh-pages instance, drag and drop their urdf, somehow assign the pkg-dependencies and finally see their robot visualized.

I agree! I've been using it to drag and drop URDF files to see what's in them and debug problems -- granted the models I'm using have pretty simplistic folder structures.

> What do you mean by multiple files? One after the other, or a folder?

All the files have to be at once. So you can drag and drop multiple files (the URDF and associated mesh files) or you can drag a full folder and the first URDF found will be displayed. Once you drag and drop a new set of files the old set of files is disregarded.

> I guess by geometry you mean the meshes?

Yup!

> How is it possible to reference them correctly with the package-name<>file-path.

Because there's no package information available it just guesses based off of the name of the file and the available directory information. So it could wind up loading the incorrect mesh if there are multiple files with the same name and directory structures in different packages but I haven't come across that, yet. You can see how it reads and loads the dragged files [here](https://github.com/gkjohnson/urdf-loaders/blob/master/javascript/example/dragAndDrop.js#L109).

> Actually, I tried today to add a pop-up that would ask for all missing package-references (e.g. raw.githubusercontent link ).

I'd considered either this or allowing you to drag and drop sets of files in succession to "build up" all the files needed to load the URDF but the single drag and drop was good enough for my use case. If you're interested in expanding that capability let's talk about what that should look like in another issue!

gkjohnson, 2019-02-15:
So I tracked down the discrepancy and it looks like the source of the issue is that there are two `origin` tags listed in the `right_front_wheel` links' visual node:

```xml
   <link name="right_front_wheel">
     <visual>

       <!-- Here -->
       <origin rpy="1.57075 0 0" xyz="0 0 0"/>

       <geometry>
         <cylinder length="0.1" radius="0.035"/>
       </geometry>
       <material name="black"/>

       <!-- And here -->
       <origin rpy="0 0 0" xyz="0 0 0"/>

     </visual>
   </link>
```

RViz must use the first instance of the tag while this loader uses the last instance. I'm not sure if it feels right to fix anything because technically the URDF file is improperly formed. Do you have any thoughts?

ipa-jfh, 2019-02-18:
Oh I am very sorry for that! Did not see that. I also do not see the need to fix anything ;)

closed 2019-02-18 by ipa-jfh