gkjohnson/urdf-loaders #121: Assign urdfLoader as class method

ipa-jfh, 2019-02-27:
@gkjohnson could you assign the `urdfLoader` function as class method? 
https://github.com/gkjohnson/urdf-loaders/blob/master/javascript/src/urdf-viewer-element.js#L483

I would need the access to overwrite it:

```js
const meshOnlyLoader = this.urdfLoader.defaultMeshLoader.bind(this.urdfLoader); 
this.urdfLoader.defaultMeshLoader = (path, ext, done) => {

    meshOnlyLoader(path, ext, obj => {
        for (let i = obj.children.length - 1; i >= 0; i--) {

            if (obj.children[i].type !== 'Mesh') {
                obj.remove(obj.children[i]);
            }
        }

        done(obj);
    });
};
```

Related to https://github.com/gkjohnson/urdf-loaders/pull/71

gkjohnson, 2019-02-27:
Sorry I removed that in the last update, but you should be able to use the `meshLoadFunc` field on the element to the same effect. That function gets used to load the models when loading the URDF. I use it in the example here:

https://github.com/gkjohnson/urdf-loaders/blob/master/javascript/example/index.js#L189

Sorry it looks like I forgot to update the docs with those functions! Note that the API for the load mesh function changed a bit, as well, and now takes `(url, manager, done)` as arguments.

I'm open to feedback on the form or names of the functions and new API -- let me know!

ipa-jfh, 2019-02-28:
Thank you for the info.

I tried to do it with the  `meshLoadFunc`. I works, but I had to copy a lot of the `defaultMeshLoader`. 
Or is there a better way how to do this?

```js
import { STLLoader } from 'three/examples/js/loaders/STLLoader';
import { ColladaLoader } from 'three/examples/js/loaders/ColladaLoader';

function meshOnlyLoader(path, manager, done) {
    if (/\.stl$/i.test(path)) {

        const loader = new STLLoader(manager);
        loader.load(path, geom => {
            const mesh = new THREE.Mesh(geom, new THREE.MeshPhongMaterial());
            done(mesh);
        });

    } else if (/\.dae$/i.test(path)) {

        const loader = new ColladaLoader(manager);
        loader.load(path, dae => {
            for (let i = dae.scene.children.length - 1; i >= 0; i--) {
                if (dae.scene.children[i].type !== 'Mesh') {
                    dae.scene.remove(dae.scene.children[i]);
                }
            }
            done(dae.scene)
        });

    } else {
        console.warn(`URDFLoader: Could not load model at ${path}.\nNo loader available`);
    }
}
...

this.loadMeshFunc = (path, manager, done) => meshOnlyLoader(path, manager, done)

```

gkjohnson, 2019-02-28:
Oh I see, you're right -- the `defaultMeshLoader` isn't all that accessible at the moment. Thanks for the use case I'll track that in another issue and think of how to address it. Moving the loader back to the class may be the quickest option.

What's the goal with your `meshOnlyLoader` function? Depending on what you are trying to achieve you could possible use the ["urdf-processed" event](https://github.com/gkjohnson/urdf-loaders/blob/master/javascript/src/urdf-viewer-element.js#L470) that fires when the model has been loaded and added to the scene. If you're trying to remove a specific class of object from the model:

```js
viewer.addEventListener('urdf-processed', () => {
    viewer.robot.traverse(c => {
        // remove certain objects here.
    });   
});
```

It's not exactly the same but maybe it'll do. I'll think about the other options this weekend.

ipa-jfh, 2019-03-11:
The goal is to remove e.g. lights from the `stl` file because they increase the loading time of the cad files by more than 200% and they overexpose the 3D scene.

Because of the first point I think that the "urdf-processed" event would be too late since one would load the objects first and then would have to remove the lights in a second step.

gkjohnson, 2019-03-11:
> The goal is to remove e.g. lights from the `stl` file

I assume you mean Collada file, right? My understanding is that STL files cannot include lights.

I recall discussing this with you before and it looks like I actually advised you to use the use this approach [here](https://github.com/gkjohnson/urdf-loaders/pull/71#issuecomment-418928811). Sorry for pulling the rug out from under you! I've been changing the API around a bit so I appreciate the feedback!

> they increase the loading time of the cad files by more than 200% and they overexpose the 3D scene.

There must be something else going on if including the lights causes a longer load time. All the lights and other components are going to be parsed and instantiated by the `ColladaLoader` by the time your function is called, anyway. I could see them causing lower framerate with more lights to render but the `urdf-processed` function should be called before the model is ever rendered. If you're still seeing performance issues with that approach I'd be curious to see what the exact problem is so we can fix it.

All that aside, how would adding the `defaultMeshLoader` function onto URDFLoader as a static member sound? That way you wouldn't need an instance of the loader to use it. Your code would change to something like this:

```js
 viewer.loadMeshFunc = (path, manager, done) => {
    URDFLoader.defaultMeshLoader(path, manager, obj => {
        for (let i = obj.children.length - 1; i >= 0; i--) {

            if (obj.children[i].type !== 'Mesh') {
                obj.remove(obj.children[i]);
            }
        }

        done(obj);
    });
};
```

ipa-jfh, 2019-03-14:
> I assume you mean Collada file, right? 

yes you are right.

It did work with your snippet. Thank you! 

Probably the longer "loading" time that I experienced was during rendering and since the lights are removed "early enough" it still loads fast with your code snippet.

 Here the final code that worked for me:
```js

import URDFLoader from 'urdf-loader/src/URDFLoader.js';

this.loadMeshFunc = (path, manager, done) => {
    new URDFLoader(manager).defaultMeshLoader(path, manager, obj => {
        for (let i = obj.children.length - 1; i >= 0; i--) {

            if (obj.children[i].type !== 'Mesh') {
                obj.remove(obj.children[i]);
            }
        }

        done(obj);
    });
};
```

closed 2019-12-25 by gkjohnson