gkjohnson/urdf-loaders #142: Consider moving to Typescript

gkjohnson, 2019-07-02:


jmcmahon443, 2023-12-05:
I am loading the ATHLETE URDF and rendering the base link geometry in react-three-fiber, the TypeScript version of three.js. However, I am not sure how to loop through the URDF file and parse it into a JSX element, similar to gltfjsx (https://github.com/pmndrs/gltfjsx).

<script src="https://gist.github.com/jmcmahon443/6c635eb2dcb35134f92370d31e91e293.js"></script>

jmcmahon443, 2023-12-05:
> I am loading the ATHLETE URDF and rendering the base link geometry in react-three-fiber, the TypeScript version of three.js. However, I am not sure how to loop through the URDF file and parse it [...]

Here's what I'd like to avoid:
```
const hip1Mesh = model.children[1].children[0].children[0].children[0] as THREE.Mesh
const hip1Joint = model.children[1].children[0].children[1] as URDFJoint
...
<mesh geometry={ hip1?.geometry }
      position={ hip1?.position }
      rotation={ hip1?.rotation }>
</mesh>
```

gkjohnson, 2023-12-07:
> react-three-fiber, the TypeScript version of three.js.

React-three-fiber is not a typescript version of three.js - it's a react integration. But generally if you need help with integrating vanilla libraries with r3f you'll have to ask at dedicated r3f forums.

Overall I'm not interested in updating the project to typescript, though, so I'll be closing this.

closed 2023-12-07 by gkjohnson