gkjohnson/urdf-loaders #149: Support for `reloadLink` and `clearLink`

tocttou, 2019-09-23:
Hello

I am looking into implementing these two functionalities:

`reloadLink(robotModel, linkNode)` : reloads the given link for the given `robotModel`, adds link if does not exist.

`clearLink(robotModel, linkNode)` : removes the link from the robot model, remove all if no link supplied.

The problem I am facing is that since all the `process*` functions are scoped to the `parse` function, the only way to implement these is through the `options` param in the `parse` function. It appears to me that it is better to instead pull up these functions as class methods for more flexibility.

Please let me know if this is suitable, or if you have another way in which these functions can be implemented. Thanks.

gkjohnson, 2019-09-23:
Can you elaborate on what your use case is a bit more so I can understand exactly what you're trying to do? Why are you adding and removing links and joints? Are you creating an editor for robot models?

Would exporting those functions separately (I don't think they have any dependency on URDFLoader? I'll have to look again when I get a chance) be suitable:

```js
export { URDFLoader, processJoint, processLink };
```

And you are expecting to parent the joints and hook them up to the `URDFRobot` class yourself, correct?

gkjohnson, 2019-09-25:
Got it! Maybe there's something simpler we can do address this? A few more questions / comments:

> We have a requirement that if a particular link fails to load for whatever reason, the user should be able to retry partial load for that link only.

So this case might be if a piece of geometry like an STL model fails to download or parse, for example? Or the geometry being served is changed or updated? The link information would be the same in this case so you'd just want to try reloading the STL file, right?

> Also a user cannot remove a robot already added to scene and switching between visual and collision will be easier if load can be done on a per link basis for meshes.

Hmm it might be worth making this easier. I see it's not in the documentation but we do have a [URDFCollider](https://github.com/gkjohnson/urdf-loaders/blob/master/javascript/src/URDFClasses.d.ts#L3-L7) type with an `isURDFCollider` field that can be used determine whether or not an object is a collision volume. Here's what I might suggest:

```js
function toggleVisibility( robot, enableColliders, enableMeshes ) {

  for ( const key in robot.links ) {
    
    const link = robot.links[ key ];
    link.children.forEach( c => {
    
      if ( ! c.isURDFJoint ) {

        if ( c.isURDFCollider ) {

          c.visible = enableColliders;

        } else {

          c.visible = enableMeshes;

        }

      }

    } );

  }

}
```

> Currently, if I understand correctly, the way to do it is to use `load` again - which triggers it for the whole urdf.

Right now that's true and for the moment I think this is what I might suggest if the robot changes but I could be convinced otherwise. You could also use `parse` and a cache downloaded geometry if you want to avoid redownloading the URDF file and processing associated geometry. This is how I'm using the loader at the moment and updating the robot if something changes and caching that data keeps the reload really fast.

gkjohnson, 2019-10-24:
Hey @tocttou! I just wanted to check in to see if you were still interested in this or if you were able to work with some of the suggestions I provided to afford reloading the robot model? If were able to get everything working as is I'll go ahead and close this.

Thanks again!

tocttou, 2019-10-25:
@gkjohnson sorry some other work has come up for me in the meanwhile. This can be closed! Thanks.

closed 2019-10-25 by tocttou