gkjohnson/urdf-loaders #150: Inverse Kinematics (IK) of Joints

kavikode, 2019-10-11:
Your work is amazing! Forward kinematics works well.

Only needed functionality is inverse kinematics. I'd like to put a target object in the scene and the end-effector should compute the IK and point in their target object. Can you please help?

gkjohnson, 2019-10-12:
Hi @kavikode! Always glad to hear about interest in the library!

Regarding IK -- it's typically a very application-specific consideration so it really depends on what you need. For our applications, for example, the kinematics solution are usually run offline or by a simulation and results are provided to the visualization client. If you need interactive IK and you need it to match you robotic system you might port the IK solution to the Javascript for consistent results.

If you just want a generic iterative IK solver there are a couple solutions available for three.js that could probably be used for the URDF model, though I haven't done so myself:

- https://github.com/jsantell/THREE.IK
- https://github.com/lo-th/fullik

I'm not sure if they encompass all of the joint constraints that URDF affords, though, so some experimentation would be needed. And of course I don't believe either solution supports closed-loop kinematic joint chains if that's required.

Depending on how complicated your kinematic system is, though, it could be pretty simple to point a joint or two in a particular direction.

kavikode, 2019-10-13:
Thank you so much @gkjohnson Can you please provide an example of adding either THREE.IK or fullik to your own ATHLETE URDF File so I can use it as a reference? It would be so helpful.
Also, which file can be updated to add different objects to the scene and zooming the camera in and out for a larger perspective?
Thank you so much for your time and I really appreciate!

kavikode, 2019-12-06:
Is there a script tag for the URDFLoader class instead of npm please? I cannot find it.

closed 2020-10-09 by gkjohnson

vantony1, 2023-04-13:
I can't figure out how to get forward kinematics to work here. I want to move from a joint configuration A to configuration B. Could you please point me in the right direction?