gkjohnson/urdf-loaders #153: Transparent and Invisible Model Robot

kavikode, 2019-12-13:
[baxter.zip](https://github.com/gkjohnson/urdf-loaders/files/3960820/baxter.zip)

Even though there are no errors after uploading the attached files for the Baxter robot model, it is not visible in the scene. It is transparent.

I turned off the shadow and it still does not work. The same problem occurs with the PR2 robot model. 
[image]
[image]

Can you please help me?

gkjohnson, 2019-12-16:
@kavikode It doesn't look like this is a problem with the URDFLoader -- please make sure your geometry loads correctly with the appropriate three.js geometry loader (STLLoader or ColladaLoader)

kavikode, 2019-12-17:
Thank you @gkjohnson I understand that I have to put all the meshes in the local file directory and remove the word "package" from the URDF file. Please see attached the updated version of the Baxter. Now, the only challenge is the PR2 even after updating the files it is the same issue.  I got those files from https://github.com/PR2/pr2_common/tree/melodic-devel/pr2_description and converted the xacro file to pure urdf file. Do you have an idea why it may not work 
[image]
and how to fix it please?

gkjohnson, 2019-12-19:
> I understand that I have to put all the meshes in the local file directory and remove the word "package" from the URDF file.

This shouldn't be necessary.

Like I mentioned I don't believe this is a problem with URDFLoader -- please check to see if the models load individually using three.js' ColladaLoader. You can see an example for how to set that up here:

https://threejs.org/examples/?q=collada#webgl_loader_collada

gkjohnson, 2020-01-06:
I'll close this for now. If it looks like the geometry loading issue is a problem with the URDFLoader we can revisit. Thanks!

closed 2020-01-06 by gkjohnson

kavikode, 2020-01-06:
how do you think we can resolve this issue for the PR2 URDF because it loads well in robot web tools?

gkjohnson, 2020-01-06:
I've taken a moment to look into both models and I maintain that these are not issues with the URDFLoader but it looks like there are separate problems with the models you posted.

Regarding the Baxter model:
- The mesh paths in the `.urdf` file are incorrect and point to the wrong folder. The paths refer to `../meshes/` while the folder sits next to the URDF file. Changing the paths to `./meshes/` allows the geometry to load.
- The Baxter model does not render when shadows are enabled because there are many PointLights embedded in the Collada files that get loaded. Removing these lights allows the shadow to render. I haven't taken the time to understand why this is a problem but it's possible that this is a bug with three.js and if so an issue should be filed in that repo.

Regarding the PR2 model:
- The PR2 geometry is invisible because the Collada models refer to TIFF image files which can't be loaded in three.js, as mentioned in [this forum post](https://discourse.threejs.org/t/i-want-to-set-the-texture-tif-file-to-the-loaded-fbx/8161). It would probably be best if an error was logged or something other than a transparent mesh was rendered. But again this should be submitted as a bug to the three.js repo.

Hopefully that helps with these models. I'm happy to help with problems and feature requests but I expect some reciprocated debugging effort. I suggested previously that you try to load the transparent meshes independently which should have made it clear that the issue with the PR2 model was elsewhere so I ask that you follow through or respond to suggestions if you're expecting help. Unfortunately I can't be asked to debug problems with other code bases because I don't have the time or energy.