gkjohnson/urdf-loaders #157: Joints Control Menu

kavikode, 2020-01-24:
[baxter.zip](https://github.com/gkjohnson/urdf-loaders/files/4106430/baxter.zip)
[fetch.zip](https://github.com/gkjohnson/urdf-loaders/files/4106434/fetch.zip)

[image]
[image]
[image]
[image]

Although the Pepper, PR2, Baxter, and Fetch models are visible now, all of the joints that have the ability to move do not appear in the joint control menu. 

Please see attached the images of the joints for each of the model and the URDF files for each model.

Can you please help me understand how I can resolve this so the joints are shown in the control menu?

http://csee.umbc.edu/~kavi1/pr2.zip
[pepper.zip](https://github.com/gkjohnson/urdf-loaders/files/4106428/pepper.zip)

kavikode, 2020-01-24:
Many thanks in advance @gkjohnson I'm grateful

gkjohnson, 2020-01-30:
It looks like if you add the css property `overflow:hidden` to the `#controls` menu it enables the scroll bar -- something may have changed with the flex styles at some point and broke this. Feel free to make a PR!

kavikode, 2020-01-30:
Thanks a million @gkjohnson You are wonderful! I'll be using your repo for my research and citing your name. Hope it is okay.

closed 2020-03-13 by gkjohnson