gkjohnson/urdf-loaders #172: Separate URDF XML Parsing

gkjohnson, 2020-06-25:
Separate the URDF XML parsing to provide a clean, fully defaulted json representation of the URDF data so it can be used to implement loaders in other engines including Babylon.js. The XML parsing is not so complicated so it might not buy a whole of savings but it's something.

The structure from the (3DTilesRenderer)[https://github.com/NASA-AMMOS/3DTilesRendererJS/] could be reused -- ie a `base` and `three` folder to denote reusable and platform-specific code.

Related to #168.

jhurliman, 2021-07-10:
I'm interested in helping with this since I'd like to use the URDF loader with Worldview (https://webviz.io/worldview/) instead of three.js. I noticed that three.js math types such as Vector3 and Quaternion are used during loading, what would you recommend in place of those for removing the three.js dependency from the core loader?

gkjohnson, 2021-07-11:
Hello @jhurliman! If the parser is going to be restructured to separate some of the parser logic I think I'd like to see a `URDFLoaderBase` implementation that just implements the XML parsing and returns a simpler JS object with defaults filled in and file paths to load resolved so rendering-engine-specific implementations can be simplified. The returned object could be structured like so:

```js
{
  namedMaterials: [ /* material data ... */ ],
  links: [ /* link data ... */ ],
  joints: [ /* joint data ... */ ],
  // ... etc
}
```

And the current URDFLoader can be split into two files:

```js
// src/base/URDFLoaderBase.js
class URDFLoaderBase {
  load( url ) {
    /* ... */
  }

  // returns parsed URDF XML as javascript object
  parse( xmlOrString ) {
    /* ... */
  }  
}

// src/three/URDFLoader.js
class URDFLoader extends URDFLoaderBase {
  parse( ...args ) {
    const urdfData = super.parse( ...args );
    /* ... generate three.js URDFRobot, etc */
  }
}

```

See the 3DTilesRendererJS [B3DMLoaderBase](https://github.com/NASA-AMMOS/3DTilesRendererJS/blob/master/src/base/B3DMLoaderBase.js) and three.js-specific [B3DMLoader](https://github.com/NASA-AMMOS/3DTilesRendererJS/blob/master/src/three/B3DMLoader.js) for a rough model of how I'm imagining this being structured. To that end I don't think any math functions will be required in the base parser and will just be required for engine-specific implementations which is probably for the best so we can keep the dependency list low.

With that approach I don't believe the result of the current URDFLoader should have to change and it should make it easier to produce other engine-specific urdf model loader implementations.