gkjohnson/urdf-loaders #184: Add support for <Mimic> tags for Joints

gkjohnson, 2020-10-10:
http://wiki.ros.org/urdf/XML/joint

stevegolton, 2021-01-30:
I'm happy to take a look at this if no one else is already?

gkjohnson, 2021-01-30:
Hey @stevegolton! As far as I know no one is working on this -- it would be great if you're interested in contributing!

I haven't thought through how this should be implemented. What were you thinking? Do you have a file that can be used for testing?

Thanks!

stevegolton, 2021-01-30:
Hi @gkjohnson,

Thanks for the fast response. I've had a very brief look at this already. The [URDF for the Robotiq 2F gripper](https://github.com/ros-industrial/robotiq/blob/kinetic-devel/robotiq_2f_85_gripper_visualization/urdf/robotiq_arg2f_85_model_macro.xacro) can be used for testing as it is mostly mimic joints, though the multipliers and offsets are not too exciting.

I hacked together a very simple example [here](https://github.com/stevegolton/urdf-loaders/commit/b7cd5a4694096bc182f6173fe7bd91fd42731150) but it's pretty inefficient and only works when setting the joint angles through the robot, rather than the joint directly.

I'm not quite sure of the best solution, but here are some ideas:

1. To each joint, add an array of references to any mimicking joints. In `setJointValue()`, also set the angle of any joints in this array.
2. Give each joint a reference to its parent `URDFRobot`, and an array of string names of any mimicking joints. In `setJointValue()` look up the joint in the robot's `joints` map, and set their angles appropriately.

Neither of the above disallow mimicking joints from having their angles set independently. Maybe this isn't an issue? Perhaps mimicking joints throw an exception in their `setJointvalue()`, or jjust do nothing.
Unsure of where the `multiplier` and `offset` should be stored: on the mimicking joint, or the mimicked joint.

I've most likely overlooked something, I'll be glad to hear any feedback you have.

Thanks!

gkjohnson, 2021-02-01:
I think it the mimicked joints should keep up to date when `URDFJoint.setJointValue()` is called rather than just when it's called on URDFRobot, so I'm leaning towards approach 1. And I'm not sure how it works in ROS but I'd expect that when a mimicked joint has `setJointValue` called on it it should update the joint it's mimicking so everything stays in sync rather than throwing an error. Does that sound right?

I'm imagining something like the follow changes:

```js
class URDFJoint extends URDFBase {

    mimicJoints: Array<URDFJoint>; // set of joints that mimic this joint

}
```

And for mimic joints make a new class like so:

```js
class URDFMimicJoint extends URDFJoint {

    offset: Number;
    multiplier: Number;
    mimickedJoint: URDFJoint;
    setJointValue( v ) {

        // TODO: apply inverted offset and multiplier
        this.mimickedJoint.setJointValue( v );

    }

    // called by the the mimicked joint when it changes
    updateFromMimickedJoint() {

        // TODO: apply offset and multiplier
        super.setJointValue( ...this.mimickedJoint.jointValue );

    }

}
```

What do you think? The other things to keep in mind is that we'll want to update the `URDFRobot.clone()` function to account for mimic joints and add a test or two to make sure it works properly.

stevegolton, 2021-02-02:
Sorry about the delay in getting back to you. I think your solution looks great. However, I took some time to poke around and ROS and see how mimic joints are handled there. As far as I know, the main handler of mimic joints is the [robot_state_publisher](https://github.com/ros/robot_state_publisher). 
It turns out if you specify the a mimicking joint's position (this is not usually expected to be done), it doesn't update the mimicked joint, and just applies the joint position to the mimicking joint.

You can see [here](https://github.com/ros/robot_state_publisher/blob/ebe1927e243c3fe84f29c164b70c87678e8e9b74/src/joint_state_listener.cpp#L149) that the mimic value is inserted in a map using `std::map::insert()`, which doesn't override existing values.

The following example demonstrates the behavior using the Robotiq gripper. All non-fixed joints should mimic the `finger_joint` joint. I am running the `robot_state_publisher` and sending `joint_state` messages to it using the `rostopic` command. Screenshots are from rviz.

From the robotiq package, I removed the `joint_state_publisher` from the following launch file and ran it like so:
```bash
roslaunch robotiq_2f_85_gripper_visualization test_2f_85_model.launch
```

Set the `finger_joint` position to 0 - the gripper is open fully.
```bash
rostopic pub /joint_states sensor_msgs/JointState "header:
  seq: 0
  stamp: now
  frame_id: ''
name: ['finger_joint']
position: [0]
velocity: [0]
effort: [0]" 
```
[image]

Set the finger_joint to 0.8 - the gripper is fully closed. 
```bash
$ rostopic pub /joint_states sensor_msgs/JointState "header:
  seq: 0
  stamp: now
  frame_id: ''
name: ['finger_joint']
position: [0]
velocity: [0]
effort: [0]" 
publishing and latching message. Press ctrl-C to terminate
```
[image]

The `left_inner_knuckle_joint` joint should mimic `finger_joint` but if the value is specified separately like so, then the angle is used regardless, and the model looks... a little broken.
```bash
$ rostopic pub /joint_states sensor_msgs/JointState "header:
  seq: 0
  stamp: now
  frame_id: ''
name: ['left_inner_knuckle_joint']
position: [0]
velocity: [0]
effort: [0]" 
publishing and latching message. Press ctrl-C to terminate
```
[image]

Even if we specify both angles in the same message, the supplied joint angle is used for this joint regardless, and the mimic is ignored. The gripper looks the same as above, and it doesn't matter which order the joints are supplied either.
```bash
$ rostopic pub /joint_states sensor_msgs/JointState "header:
  seq: 0
  stamp: now
  frame_id: ''
name: ['finger_joint', 'left_inner_knuckle_joint']
position: [0.8, 0]
velocity: [0]
effort: [0]"
```

I'm not sure whether this behavior is intentional or not, and I'm sure this can be improved on. What are your thoughts? Should we stick with the ROS way, or try to improve on it using what you suggested above?

One more thing to be wary of is infinite loops of mimicking, but I guess this could be caught at load time.

gkjohnson, 2021-02-03:
Thanks for putting in the research!

> Should we stick with the ROS way, or try to improve on it using what you suggested above?

In that case why don't we just log a warning if a mimic joint is set directly instead of through the mimicked joint and stick with the way ROS handles it. I'm not using mimic nodes in my work so I don't have strong opinions but I figure this can be adjusted in the future if needed if that sounds good to you.

> One more thing to be wary of is infinite loops of mimicking, but I guess this could be caught at load time.

I would expect this could only happen if the URDF defines a "ring" of connected mimic joints, which seems invalid. But if we do move towards the approach I mentioned above we could validate the structure on load.

closed 2021-04-16 by gkjohnson