gkjohnson/urdf-loaders #198: Remove dependency on THREE.js?

jtbandes, 2021-04-15:
Thanks for publishing this package! I'm looking at integrating it into a project that doesn't currently use THREE.js for 3D rendering. I was interested in just using the URDF parsing parts, but since the package uses the Quaternion/Euler classes from three, I didn't see a way to do that. It would be great if the dependency was optional, or perhaps if the parsing were split out into its own package (maybe using something like gl-matrix for a smaller math library).

gkjohnson, 2021-04-16:
Thanks for your interest! Three.js is used pretty extensively in the `URDFLoader` component between materials and instantiating joints and links so it's a bit more than just the quaternion and euler classes but others have asked in the past for something more agnostic.

I wouldn't be opposed to something like a `URDFParser` or `URDFLoaderBase` class that contains all the rendering engine-agnostic bits and returns something like simplified JSON representation of the joints, links, etc in the URDF that can then be instantiated with engine-specific set of objects. In that case I don't think `gl-matrix` would be needed because any math or rotation application can be deferred to the full instantiation stage.

Unfortunately I don't have the bandwidth to put the effort into refactoring right now, though. If you're interested in taking something like that on I'd be happy to talk it through and refine a plan and help get a PR going to get it merged.

jtbandes, 2021-04-16:
Yep, a plain parser that just returns a JSON-like object is what I was thinking (something like this simple [GLB parser](https://github.com/cruise-automation/webviz/blob/7407ef1687e19615a43194c003aec6608c4f7c51/packages/regl-worldview/src/utils/parseGLB.js#L21) I worked on a couple years ago – although that's simpler because a large part of the data format is already json).

Makes sense that this might be a bit of work to extract from the current code structure — for now, I'm going to try integrating three.js since we might eventually be interested in moving over the rest of the project as well. Thanks for the quick response!

gkjohnson, 2021-04-16:
Sounds good -- I'm going to close this as a dupe of #173 for now, then. 

Are you planning to use this for Cruise Automation projects / Webviz, then? I've used your Rosbagjs library in the past which has been very useful (I knew I recognized your icon from somewhere). Not sure if you're looking into parsing Xacro files as well as URDF but I've published a js xacro-parser package [here](https://github.com/gkjohnson/xacro-parser) that can be used to preprocess any Xacro URDF files before loading with this package.

closed 2021-04-16 by gkjohnson

jtbandes, 2021-04-16:
Yep, I remember some of our conversations on rosbag.js back in the day 😄 I really appreciate the work you've done on these packages and being responsive to PRs!

I'm now working on [Foxglove Studio](https://github.com/foxglove/studio), which started as a fork of Webviz, but is now a desktop app which allows for things like filesystem access and direct TCP connections to ROS graphs. URDF support is a common request!

The app has a lot of features and inertia in our current 3D panel, but we're open to / interested in eventually migrating to three.js — it'll just take a while. For now the URDF viewer will probably be a separate panel so we can get something working sooner, and visualizing the URDF alongside other markers and TFs and such will come later.