gkjohnson/urdf-loaders #199: TypeScript definitions aren't shipped alongside umd sources

jtbandes, 2021-04-15:
When I import `urdf-loader` into a TypeScript project (with webpack, and VS Code as an editor), it uses the sources from the `umd` directory. But since the .d.ts files are only present in the `src` directory, the type definitions aren't automatically loaded. I think shipping .d.ts alongside the .js files in the umd folder would solve this problem.

It would also be nice to export all the URDFClasses from the main source file, so that you can do e.g. `import URDFLoader, { URDFRobot } from "urdf-loader"`.

gkjohnson, 2021-04-16:
Hello! I've been slowly moving packages I maintain to use `type: "module"` to avoid having to deal with these types of issues. I've made PR #200 which adds the field to package.json. This should force typescript to resolve to the files in src and solve the types resolution issue. Let me know if that sounds okay.

> It would also be nice to export all the URDFClasses from the main source file, so that you can do e.g. `import URDFLoader, { URDFRobot } from "urdf-loader"`.

Can you elaborate on why you want access to the URDFRobot, URDFJoint, etc classes? They have not really been designed to instantiate individually by a user. At least not without some extra effort.

jtbandes, 2021-04-16:
> Can you elaborate on why you want access to the URDFRobot, URDFJoint, etc classes? They have not really been designed to instantiate individually by a user. At least not without some extra effort.

That makes sense. I wasn't looking to instantiate them directly, but wanted to access the type definitions so I could, for example, store a `type State = { robots: URDFRobot[] }`.

gkjohnson, 2021-04-16:
> That makes sense. I wasn't looking to instantiate them directly, but wanted to access the type definitions so I could, for example, store a type State = { robots: URDFRobot[] }.

Got it -- I'm definitely not a typescript expert but is it possible to export only the types in a d.ts file and not export the actual class? I guess this would just be something like an `interface`. Perhaps it's as simple as changing `class`  to `interface` in the d.ts files?

jtbandes, 2021-04-16:
I think you're right that changing it to an `interface` would work, since an interface can only be used as a type, not a value (you can't do `new SomeInterface()`).

gkjohnson, 2021-04-16:
> I think you're right that changing it to an interface would work

Just made PR #202 to convert classes to interfaces. I'll let you know when I've published it in the next day or so if you could let me know if something is amiss in the type imports when you use it that would be appreciated!

closed 2021-04-16 by gkjohnson

jtbandes, 2021-04-16:
I gave this a try in v0.10 and it works! But I still have to import the classes separately, like:
```ts
import URDFLoader from "urdf-loader";
import { URDFRobot } from "urdf-loader/src/URDFClasses";
```
It would be nice to just export them from src/index.d.ts as well.

gkjohnson, 2021-04-16:
Thanks for testing -- can you confirm that exporting the classes from from `src/URDFLoader.d.ts` fixes the issue? If so we can just reexport the interface definitions from there, too.

jtbandes, 2021-04-17:
Yep, adding `export * from './URDFClasses';` to URDFLoader.d.ts seems to work as expected.

gkjohnson, 2021-04-17:
Great! Just published version v0.10.1 that adds that line to `URDFLoader.d.ts`.

PR #200 (2021-04-16): Convert package to use type: module
Fix #199
Related to #169