gkjohnson/urdf-loaders #205: Copying URDFRobot

jtbandes, 2021-04-20:
I'd like to clone a URDFRobot so I can potentially use it in multiple three.js renderers at the same time. It looks like there's a `.copy()` method but it's not exposed in the TypeScript definitions, and even if it were, I wouldn't be able to `new URDFRobot()` because of #202. Is copy() safe to use for this purpose? How would you feel about exposing the class to TypeScript for this purpose, or maybe adding a `clone(): URDFRobot` method?

jtbandes, 2021-04-20:
I found a partial workaround with `new (model.constructor as any)().copy(model, true)` — this allows me to add the model to multiple scenes, but the materials are not cloned, so any changes I make (like setting opacity) apply in all places. I suppose I can also manually traverse and clone the materials.

gkjohnson, 2021-04-21:
> I learned that `clone()` already works

Yes clone is inherited from [Object3D](https://threejs.org/docs/index.html#api/en/core/Object3D.clone). which is the base class of `URDFRobot`. It's been overridden to correctly populate all the `joints`, `links`, etc fields with the appropriate new children when cloning, as well.

> the materials are not cloned, so any changes I make (like setting opacity) apply in all places. I suppose I can also manually traverse and clone the materials.

Both the material and geometry references are retained in the clone so if you want to change the materials separately on the cloned robot you'll have to traverse over the children and clone them. That behavior is a side effect of calling `Mesh.clone` and I'd not to be inconsistent with it.

It should be a pretty short call to clone the materials, though:

```js
robot.traverse( c => c.material && ( c.material = c.material.clone() ) );
```

jtbandes, 2021-04-22:
Thanks, that makes sense. I also found some related issues explaining this property of meshes:

https://stackoverflow.com/a/29580441/23649
https://github.com/mrdoob/three.js/issues/14223

closed 2021-04-22 by jtbandes