gkjohnson/urdf-loaders #210: Joint axes misaligned in global space

ndahn, 2021-05-30:
I found a very strange issue where joint axes are not aligned anymore when transformed to global space. Do you have by chance any idea where this is coming from? 

 	scene.add(robot.model);

	for (let joint of robot.arm.movable) {
		let pos = joint.getWorldPosition();
		let jglobal = joint.localToWorld(joint.axis.clone()).normalize();
		// local axis, teal, perfectly fine
		joint.add(new ArrowHelper(joint.axis, new Vector3(), 0.35, 0x00ffff));
		// global axis, pink, misaligned
		scene.add(new ArrowHelper(jglobal, pos, 5, 0xff00aa));
	}

[image]

gkjohnson, 2021-05-30:
Hello! [Object3D.localToWorld](https://threejs.org/docs/?q=object#api/en/core/Object3D.localToWorld) will transform a _position_ from the local frame into the world frame where as the joint axis represents a direction vector. Look into [Vector3.transformDirection](https://threejs.org/docs/?q=vect#api/en/math/Vector3.transformDirection), which should do what you want:

```js
jglobal = joint.axis.clone().transformDirection( joint.matrixWorld );
```

Of course make sure `joint.matrixWorld` is up to date, as well.

ndahn, 2021-05-30:
Ha! This is embarassing :D Thank you a lot!

closed 2021-05-30 by ndahn