gkjohnson/urdf-loaders #220: Exporting URDF's out of Scene

ugurkoysuren, 2021-12-06:
I'm currently trying to export an URDFRobot out of the Scene using the obj.toJSON method but when I load it back using the ObjectLoader all Types are missing from URDFRobot until all URDFLinks. They are now all Object3D. Is it possible to export a URDFRobot from Scene easily?

gkjohnson, 2021-12-06:
This is a limitation of the three.js ObjectLoader and format -- it can only export and load native three.js objects. Exporting a URDF from a 3d hierarchy is complicated and they're usually created by hand or in other tools. I have worked on an exporter in the past but it has issues and I'm not currently working on or maintaining an exporter or editor at the moment.

ugurkoysuren, 2021-12-09:
Thanks for the reply!

closed 2021-12-09 by gkjohnson