gkjohnson/urdf-loaders #224: Convert material colors to Linear on load

gkjohnson, 2022-01-31:


EzraBrooks, 2022-05-18:
What does this entail? I've attempted to upgrade my Three version as part of my usual dependency-upgrade chores and my robots' colors are very different from how they've previously looked. Does `processMaterial` just need to load the color as RGB and then invoke `convertSRGBToLinear` on it?

closed 2022-05-18 by gkjohnson

EzraBrooks, 2022-05-18:
Is this worth cutting a release for, or should I implement a workaround in my application in the `loadMeshCb` for now?

gkjohnson, 2022-05-18:
Was just getting to it 😁 Should be published in v0.10.4

EzraBrooks, 2022-05-18:
Excellent! Thanks for being so responsive!

EzraBrooks, 2022-10-17:
I'm still having some trouble with this, and it's very odd. If I use `npm link` to use a local copy of 0.10.4 in my project, my colors look fine. But using the package straight from the NPM registry still produces incorrect colors. I'm baffled 😕

gkjohnson, 2022-10-17:
I would make sure v0.10.4 is actually installed. This line here is the important one for converting the colors, as well - so you can set breakpoints to make sure it's getting here when you expect:

https://github.com/gkjohnson/urdf-loaders/blob/8f4ae740a3d7e06e810208e78105bb9592d0883b/javascript/src/URDFLoader.js#L476

EzraBrooks, 2022-10-18:
I'll check back in later today after debugging further, but my current hunch is that my application is resolving the import to the UMD version that hasn't been updated. I saw there's a ticket for removing that; is there anything blocking that change?

gkjohnson, 2022-10-18:
You can see in the UMD module that the converSRGBToLinear conversion is there, as well, on line 897.

https://unpkg.com/browse/urdf-loader@0.10.4/umd/URDFLoader.js

> is there anything blocking that change?

It's technically a breaking change and there are still quite a few build processes that require commonjs-style imports, unfortunately. I've tried to remove UMD on some of my other projects and it's been met with complaints.

EzraBrooks, 2022-10-18:
You're right, I was looking at the committed copy in the repo, oops. Hmm. I can definitely verify that it is v0.10.4 installed in my project, and that I can see the conversion happening as written. Breakpoints show that it is hitting those lines. Very very strange..

gkjohnson, 2022-10-18:
Hm 🤔 without a repro I don't think I'll be able to suggest anything else.

EzraBrooks, 2022-12-01:
Just to follow up: it was a bug in the COLLADA material declarations in the robot description from the manufacturer.

PR #233 (2022-05-18): Convert sRGB URDF colors to Linear
Fixes #224.