gkjohnson/urdf-loaders #226: Convert URDFRobot from interface to class in typescript definition

joaopfg, 2022-02-04:
Hello,

I see that `URDFRobot` is defined as a class in `URDFClasses.js` but it's defined as an interface in `URDFClasses.d.ts`.

It disallows the instantiation of an empty URDFRobot, which would be useful, for example, if I want to define my URDFRobot instance in a scope above where it's loaded.  

Is there some special reason to define it as interface in typescript ? I can try a pull request to correct it if it's not the case :)

gkjohnson, 2022-02-04:
The URDFRobot isn't exported as a class because it's not designed to be instantiated directly. You can set intermediate variables to "null" until the robot is complete loaded.

joaopfg, 2022-02-04:
@gkjohnson With more strict typescript compiler options, it's not possible to set it to `null` . My work around for now is to declare the object as `URDFRobot | undefined` and using type narrowing when needed. So far it's working well. But I don't know if it's good practice in typescript.

gkjohnson, 2022-02-04:
Using nullable types is valid in most languages. I'm not a typescript expert but type unions are absolutely valid in the language. It's documented here in the typescript docs here:

https://www.typescriptlang.org/docs/handbook/advanced-types.html#nullable-types

I appreciate the effort to contribute back to the project but I'd like to make sure the target environment is well understood before trying to change the library to conform to these issues.

closed 2022-02-04 by gkjohnson