gkjohnson/urdf-loaders #229: Cloning a URDFRobot results in mimic joints that still point to the original object

EzraBrooks, 2022-04-19:
We've got an application where we load a robot's URDF and then clone it for use in some motion preview features. However, we've found that if you `setJointValue` on a joint in the clone that has mimic joints, its mimics move in the original object but not the clone.

gkjohnson, 2022-04-19:
Thanks for the report! Would you like to make a PR with a fix for the problem as well as a new test or two?

EzraBrooks, 2022-04-19:
I'll take a stab at it; I'm departing on vacation this evening, but I'll see if I can get a PR in sometime next week.

EzraBrooks, 2022-05-03:
we have a workaround for this issue (loading the robot twice) that's not ideal but is keeping us going for now. I will try to circle back to this when I get a chance, but currently have a bunch of non-3D-related issues on my plate for our next few sprints 😄

PR #283 (2024-02-20): Fix mimic joints pointing to the original robot after a clone
Sometimes "I'll do it in a couple sprints" turns into "a year and a half later".

Fixes #229.

Adds an object equality test to the mimic joint tests that failed before I made the changes here.
closed 2024-02-20 by gkjohnson