gkjohnson/urdf-loaders #230: TypeScript types incorrectly exclude the `err` argument from `MeshLoadDoneFunc`

EzraBrooks, 2022-04-27:
I have an extension class in my application that does a few things, one of them being coercing `URDFLoader` into the same shape as `Loader<T>` from Three so that `react-three-fiber` is happy to call it from its `useLoader` hook. I also overrode `loadMeshCb` to allow me to call a custom function on each mesh on a per-robot basis.

While making that `loadMeshCb` override, I discovered that the `onLoad` argument of form `(obj: Object3D, err: any) => void` is typed in the `.d.ts` file as 
```
interface MeshLoadDoneFunc {
    (mesh: Object3D): void
}
```

And thus had to ignore compiler errors via `@ts-ignore` or `@ts-expect-error` to properly override the function.

EzraBrooks, 2022-04-27:
Actually, poked around in the code a little more and now I'm unsure which form is correct - seems like the types and the JS are mismatched?

EzraBrooks, 2022-04-27:
Sorry, that comment was vague 😆. I looked at `defaultMeshLoader` and it looks like it calls `done(mesh)`, with no `err` arg whatsoever. Correct me if I'm wrong, but doesn't that mean the `err` is unused and is always resolving to `undefined`?

gkjohnson, 2022-04-27:
`err` is an optional argument to the "done" callback. The default callback could probably be updated to pass an error on failure but it currently doesn't.

EzraBrooks, 2022-04-27:
Sounds good. I'll update the types file, then.

PR #231 (2022-04-29): Fix MeshLoadDoneFunc type to include err
Fixes #230.