gkjohnson/urdf-loaders #232: Using Loader with React Three Fiber?

D-Niermann, 2022-05-11:
We are currently working on a project with React (and Typescript) that uses React Three Fiber to implement a Three.js scene.
How to use the URDF Loader (and also Xarco Loader) with Three Fiber? A small example would be ideal.
Using the hook 'useLoader(URDFLoader, url)' does not work.

Thank you!

gkjohnson, 2022-05-11:
I am unfamiliar with react three fiber so I won't be able to help, unfortunately. If you figure it out feel free to contribute documentation to the project with instructions.

EzraBrooks, 2022-05-18:
I have a wrapper around `URDFLoader` that allows its use in `useLoader`. It's almost certainly naive and incomplete, but it works for my use case. [Here's a gist](https://gist.github.com/EzraBrooks/ca44740038c294ad727b702737ea82c3). All you have to do to consume this is call `useLoader` and bind the resulting `URDFRobot` to a `<primitive>` from R3F.

D-Niermann, 2022-05-23:
Thanks for the answers! 
I also got it to work with a simple but hacky //@ts-ignore flag. The ts-error was probably the only thing that got me confused.
Here is my code:
```
function loadURDF(filepath){
	//@ts-ignore
	const mesh = useLoader(URDFLoader, filepath, (loader) => {
		loader.loadMeshFunc = (path, manager, done) => {
			new STLLoader(manager).load(
			path,
			(result) => {
				const material = new THREE.MeshPhongMaterial();
				let mesh = new THREE.Mesh(result, material);
				mesh.position.x = Math.random()*3
				done(mesh);
			},
			null,
			(err) => done(null, err)
			)
		}
		})
	setTimeout(()=>{
		// put delay on traverse so that shadows are set, this delay is needed
		mesh.traverse((c) => {c.castShadow = true})
		mesh.traverse((c) => {c.receiveShadow = true})
	},2000)
	return mesh
}
```

closed 2022-08-18 by gkjohnson