gkjohnson/urdf-loaders #236: 请问在vue3里面如何使用这个urdf-loader

coderPhy, 2022-07-20:
<template>

<urdf-viewer
        ref="viewerRef"
        display-shadow
        tabindex="0"
        up="Z"
      ></urdf-viewer>

</template>

    const ExampleElement = defineCustomElement(URDFManipulator)
    customElements.define('urdf-viewer', ExampleElement)
const scene = new THREE.Scene()
    loader.loadMeshCb = function (path, manager, onComplete) {
      const stlLoader = new STLLoader(manager)
      stlLoader.load(
        path,
        (result) => {
          onComplete(result.scene)
        },
        undefined,
        (err) => {
          onComplete(null, err)
        }
      )
    }
    loader.load('../../urdf/excavator_3.urdf', (robot) => {
      // The robot is loaded!
      scene.add(robot)
    })

coderPhy, 2022-07-20:
考虑出一个vue 3.2+的案例么

gkjohnson, 2022-07-20:
You'll have to ask a vue-related help forum for information how to use web components.

closed 2022-07-20 by gkjohnson