gkjohnson/urdf-loaders #250: No visual, with no errors

nicholscrawford, 2023-01-26:
Trying to load an iiwa URDF, with .obj files. It seems like I need to provide my own loader callback, which I did like this:

```
loader.loadMeshCb = function( path, manager, onComplete ) {

        const objLoader = new OBJLoader( manager );
        objLoader.load(
            path,
            result => {
                onComplete( result.scene );
            },
            undefined,
            err => {
                // try to load again, notify user, etc
                onComplete( null, err );  
            }
        );
    };
```
And otherwise everything else is basically copied exactly from the example. The only other thing I noticed that was weird was the bounding box seems to have infinite lengths, but even setting the robot position to 0,0,0 didn't seem to fix it. That seems to suggest the geometry is not loading, but I'm not getting any errors. Is this a proper callback? Is .obj built in?

gkjohnson, 2023-01-27:
> ```js
> onComplete( result.scene );
> ```

OBJLoader does not return an object with a "scene" field. See the [OBJLoader docs](https://threejs.org/docs/index.html?q=objloa#examples/en/loaders/OBJLoader). I would make sure you're using the mesh loader correctly by using some log statements first.

nicholscrawford, 2023-01-27:
Oh thank you!

nicholscrawford, 2023-01-27:
Resolved by removing .scene. Thanks for the tips, I'm new to js.

closed 2023-01-27 by nicholscrawford