gkjohnson/urdf-loaders #255: Is it possible to use this repo as a base to code SDF loaders in javascript?

addy1997, 2023-03-27:
Hi @gkjohnson ,

Your project is really interesting. I am wondering whether it is possible to code similar loader for SDF (Simulation Description Format) files. The purpose is to visualize SDF files of 3D models and objects in VR.

gkjohnson, 2023-03-27:
Of course it is possible but that would be outside the scope of this project. The SDF format is much larger and more complicated than URDF. The specification for the SDF XML format should be available on the ROS wiki somewhere.

addy1997, 2023-03-27:
> Of course it is possible but that would be outside the scope of this project. The SDF format is much larger and more complicated than URDF. The specification for the SDF XML format should be available on the ROS wiki somewhere if you'd like to make a loader.

Yeah, thanks for your reply. While writing the URDFclasses, did you refer to any library apart from the URDF structure? I am quite curious about your approach. Sorry for the barrage of questions.

gkjohnson, 2023-03-28:
> did you refer to any library apart from the URDF structure?

No - with a well-defined format specification you shouldn't need to reference any code and the URDF spec is fairly complete. I'm not familiar with the SDF format itself so I can't say how reliable the online spec documentation might be.

closed 2023-03-28 by addy1997