gkjohnson/urdf-loaders #256: URDF file loading KO

yvanblanchard, 2023-04-05:
Hello,

I tried to load this URDF file (KUKA iiwa7_r800), but it does not seem to work properly (no 3D view). 
Tested in both Firefox and Chrome browser (using online version of tool, by drag and drop urdf file).

Could you please try yourself ?
Thank you.

[image]

[lbr_iiwa7_r800-urdf-package-master.zip](https://github.com/gkjohnson/urdf-loaders/files/11156589/lbr_iiwa7_r800-urdf-package-master.zip)

gkjohnson, 2023-04-05:
Please look at the javascript console if you're having issues:

[image]

yvanblanchard, 2023-04-05:
Thank you, but I don't have the same messages as yours: 

[image]

Do you think it's an issue related to the DAE / Collada mesh files ? if yes, what's wrong with thos files (exported from Solidworks, it seems) ?

gkjohnson, 2023-04-06:
You need to provide more information about what you're doing, then. If you make an issue please always include reproduction steps and error logs otherwise I'm just left guessing. What files are you dragging into the page?

yvanblanchard, 2023-04-06:
Ok sorry.
The scenario is:
- open your online web app (link from GitHub) in Chrome
- drag and drop the Urdf file alone (se attached zip file)
- Result: the robot name is well displayed, and number of axis as well, but nothing displayed in the 3d viewer.

Thank  you.

gkjohnson, 2023-04-06:
> drag and drop the Urdf file alone (se attached zip file)

You need to draw and drop the URDF and all associated geometry files into the browser. Otherwise the browser cannot load the relevant mesh files. In this case you can drag the full "urdf" and "mesh" folders in the page.

gkjohnson, 2023-04-06:
> I have now the same error messages as you (three.colladaLoader, invalid opaque type "null" of transparent tag).

The warning indicates that there's a problem in the three.js ColladaLoader which is not a part of this project. I'd make sure the file loads separately and ask for help at the three.js forum.

yvanblanchard, 2023-04-06:
OK, thank you again. 
Do you want me to submit an issue to ThreeJS, or do you want to directly handle it yourself ? (to avoid posting same request twice).

gkjohnson, 2023-04-07:
This is an issue you're having so you're responsible for testing whether the latest three.js fixes the ColladaLoader issue and determining whether there's a bug in the first place or if it's an issue in the file.

closed 2023-04-07 by gkjohnson

yvanblanchard, 2023-04-07:
OK thank you very for your help Garrett.

yvanblanchard, 2023-04-07:
@gkjohnson  I have an answer from threejs about this issue, and it's now closed since the DAE files related to the urdf can be easily loaded in three-js viewer...

gkjohnson, 2023-04-14:
> the DAE files related to the urdf can be easily loaded in three-js viewer...

This is why I suggested testing with the latest version of three.js first to see if the issue was still present in the loader. If you'd like to make a PR to update the project to use the latest version of three.js I'm happy to get that merged!