gkjohnson/urdf-loaders #266: Zoom issue with urdf model

yvanblanchard, 2023-05-14:
Hello,

I face troubles when trying to zoom using mouse with this model. I can't see the whole scene. I am using the online viewer on Chrome browser (drag and drop both urdf file with STL folder).

[robot-test-urdf.zip](https://github.com/gkjohnson/urdf-loaders/files/11472869/robot-test-urdf.zip)

[image]

gkjohnson, 2023-05-17:
The model you've provided is thousands of meters large.

yvanblanchard, 2023-05-17:
My bad ! the STL files were in mm
Thanks a lot

[image]

closed 2023-05-17 by yvanblanchard