gkjohnson/urdf-loaders #270: dragging in simple example

Jayce-liang, 2023-05-31:
Can the joints of the model only be dragged or moved when loaded using<urdf view></urdf view>?

If you can't achieve joint dragging like this, right?
```
 // Load robot
    const manager = new LoadingManager();
    const loader = new URDFLoader(manager);
    loader.load('../../../urdf/T12/urdf/T12_flipped.URDF', result => {
        robot = result;
    });
    // wait until all the geometry has loaded to add the model to the scene
    manager.onLoad = () => {
        scene.add(robot);
    };
```

gkjohnson, 2023-05-31:
No - the URDFLoader is just a model loader so no automatic dragging. Any interactions will have to be added yourself. You can take a look at the source of the project to see how it's done - some of it may currently be reusable.

closed 2023-05-31 by gkjohnson

Jayce-liang, 2023-06-01:
> No - the URDFLoader is just a model loader so no automatic dragging. Any interactions will have to be added yourself. The dragging in the demo is only available in the web component. You can take a look at the source of the project to see how it's done - some of it may currently be reusable.

thank youe reply. I used the webcomponent from the example in vue2 and it worked.