gkjohnson/urdf-loaders #286: Enable to load urdf files from vs code

virtuamechanica, 2024-05-05:
Hello,

i would like to run the urdf load example from vs code, but it fails loading the urdf files (samples ones). I can successfully render the main html page, seing the menus, but urdf files can’t be processed..
if I run through npm start, it works fine.

Are there well know limitations using it from vs code ?
I just want to debug and understand the code in détails Before integrating it in another app.
(Nb: I know that you shared a vs code preview extension, so I was wondering why this urdf loader repo would not work properly..)

Thank  you for your help

gkjohnson, 2024-05-06:
This project is designed as a library to be used in a project - not for loading a VS Code extension. As you've seen I have experimented with displaying a URDF as a plugin but it's not something I'm working on or supporting any future.

gkjohnson, 2024-05-06:
I don't know anything about running from VS Code but have you looked at the documentation on how to run the project examples locally?

https://github.com/gkjohnson/urdf-loaders/tree/master/javascript#running-the-example

yvanblanchard, 2024-05-06:
Thank you for your quick reply Garett.

I was finally able to solve my problem.
I just had to correctly specify the url in the 'launch.json' configuration file of VS Code:

[image]

(nb: don't use double quotes for the url path).

closed 2024-05-06 by gkjohnson