gkjohnson/urdf-loaders #287: Hexapod kinematics failed

yvanblanchard, 2024-05-12:
Hello,

I tried to load the hexapod URDF file with the loader, but the result does not seem fine:

[image]

I know that this loader does not contain IK solver (just URDF viewer), but I think components should be correctly placed and connections (joints) between them as well.

Is it a bug or a limitation of the actual loader library ?
Thank for your help.

gkjohnson, 2024-05-12:
> Is it a bug or a limitation of the actual loader library ?

No this is how the "0" orientation are defined for the joints in this URDF model.

> I think components should be correctly placed and connections (joints) between them as well.

There is no requirement in the URDF specification that the 0-angle joints be oriented in a physically possible way. When using a robotic model like this the joint values reported by the solver in the ROS system would orient the kinematics correctly.

Unfortunately URDF does not support parallel-kinematic systems so there's no way to infer from the file how these joints should be connected without some manual annotation.

yvanblanchard, 2024-05-12:
OK, clear , thank you Garett

closed 2024-05-12 by yvanblanchard

virtuamechanica, 2024-05-12:
@gkjohnson please notice that there is a trick to handle closed chains kinematics in urdf: through gazebo xml tag:
https://github.com/2b-t/closed_loop/blob/humble/urdf/closed_loop.urdf.xacro