gkjohnson/urdf-loaders #310: Loaded urdf succesfully but can't see the logo texture

DmitriyMolch, 2025-09-29:
Loaded the URDF file successfully, but can't see the logo texture. 

Could you help fix it?

[image]

[``](url)

DmitriyMolch, 2025-09-30:
Here is the model as you requested

DmitriyMolch, 2025-09-30:
[](https://drive.google.com/file/d/1HpF1K7ja4igcA2CqCK2-pAWTEdh4JOoS/view?usp=sharing)
Here is the model as you requested

DmitriyMolch, 2025-09-30:
Also, I can't implement joint rotation as joints are rotated in the opposite direction. Do you know how to fix it? Is joint rotation supported out of the box by the "urdf-loader" package?

gkjohnson, 2025-09-30:
> Here is the model as you requested

Thank you - it looks like the logo is present and loads in the model just fine. You may need to improve the lighting in your scene if you feel the look of the model is insufficient.

[image]

> I can't implement joint rotation as joints are rotated in the opposite direction

Joints are adjusted using the [functions outlined in the README](https://github.com/gkjohnson/urdf-loaders/tree/master/javascript#setjointvalue) and the rotation direction is determined using the right-hand-rule as dictated by the URDF specification and ROS coordinate system.

DmitriyMolch, 2025-10-01:
> > Here is the model as you requested
> 
> Thank you - it looks like the logo is present and loads in the model just fine. You may need to improve the lighting in your scene if you feel the look of the model is insufficient.
> 
> <img alt="Image" width="404" height="377" src="https://private-user-images.githubusercontent.com/734200/495928676-5871df89-20af-4845-9cc6-a94810caae5a.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTkzMjAzMTgsIm5iZiI6MTc1OTMyMDAxOCwicGF0aCI6Ii83MzQyMDAvNDk1OTI4Njc2LTU4NzFkZjg5LTIwYWYtNDg0NS05Y2M2LWE5NDgxMGNhYWU1YS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUxMDAxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MTAwMVQxMjAwMThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04NjVkNWM3OWQyNjJjMTUzZmI2YmMyYmYzM2FmOWNjZjYxMjZkOTdiYTJiYzY1OWYwMzEyNmFiMWE3ZWQwOWFlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.qde8TcAzMU9Ad6EyrbphdoKOv4jNqJw44W98ek_imBo">
> 
> > I can't implement joint rotation as joints are rotated in the opposite direction
> 
> Joints are adjusted using the [functions outlined in the README](https://github.com/gkjohnson/urdf-loaders/tree/master/javascript#setjointvalue) and the rotation direction is determined using the right-hand-rule as dictated by the URDF specification and ROS coordinate system.

How to improve lighting? Could you give specific steps?

gkjohnson, 2025-10-03:
The drag controls for this project are implemented in [this file](https://github.com/gkjohnson/urdf-loaders/blob/master/javascript/src/URDFDragControls.js) but unfortunately I can't otherwise provide detailed direction on the math. If you have questions on how to calculate world rotations from something like mouse movement I'd consider making a thread at the [forums](https://discourse.threejs.org/) or stack overflow to ask for help.